ArduCAM/ArduCAM_USB_Camera_Shield #120: MT9J001 USB transfer timeout error

Synapse101, 2019-09-09:
I have an MT9J001 camera connected to a windows 10 laptop via an Arducam USB 2 camera shield. I can't seem to get the USBTest demo to display a captured image. The camera init was successful but nothing happens when I click on "shot", and I get the "USB transfer timeout error" shown in the image when I click on "play". 

Please advise.
[image]

UCTRONICS, 2019-09-10:
@Synapse101 
It seems the camera does not connect normally. the id of the MT9J001 should be 0x2c01 instead of 0x85F9, please refer to this link to get the detail information
https://www.arducam.com/docs/camera-breakout-board/10mp-mt9j001-mt9j003/software-guide/

Synapse101, 2019-09-10:
Hi, thanks for the quick reply. I've followed the examples on the provided link but keep getting the same error. The drivers have been correctly installed and I've double-checked that the cable is correctly connected.  The camera code is UC-397 rev A and USB camera shield code is UC-391 rev D.
Please advise.

ArduCAM, 2019-09-10:
Please contact us with email: support@arducam.com and send us your hardware setup photos.

Synapse101, 2019-09-19:
Hi, I've sent an email to support@arducam.com as requested but am yet to receive a reply. Is there any update?