ArduCAM/ArduCAM_USB_Camera_Shield #121: IMX298 is listed as supported, but missing cpp_config

zorrobyte, 2019-09-14:
Where can I find the cpp_config for IMX298?

Can IMX298 be used without cpp_config? How?

zorrobyte, 2019-09-14:
Also no python_config. I do intend to use this camera with Python.

ArduCAM, 2019-09-15:
Would you please let me know what kind of USB camera shield and MIPI adapter board you are using.
There are different config files for different hardware.
The Windows config file for USB3+MiPi adapter board can be found from, please try them first.
http://arducam.com/downloads/modules/IMX298/config/

zorrobyte, 2019-09-15:
Thank you for the reply @ArduCAM 

I have `ArduCAM USB2 Camera Shield (Rev.E) - Support MIPI Interface Sensors` and
`Arducam IMX298 MIPI 16MP Color Camera Module for Raspberry Pi`

The goal is to get this working with a OnePlus 3t/Android for OpenPilot, a self driving car platform used by thousands of people. It uses the factory phone camera, and since it's in the sun, it overheats the phone. The phone uses IMX298 module internally, and AI model is trained on IMX298.

If I can get this working, many people will buy the Shield and IMX298.

The requirements:
Run on Android, OpenPilot uses Python. https://github.com/commaai/openpilot/blob/devel/selfdrive/visiond/cameras/camera_qcom.c#L549
At least 20 FPS over USB 2 (supported phones only have USB 2)
At least 256x512 width image (AI model uses this resolution as input tensor) but larger the better at 20FPS.

I know supporting users can be time consuming. However, if we get this working, we can sell many of your shields and camera modules as people have wanted external camera for their self driving car for a very long time.

[image]

Right now I have gotten a Logitech C270 working over USB OTG. I have yet to get Arducam working at all, or on Linux. I'm not sure if it has a Video for Linux driver.

ArduCAM, 2019-09-15:
I love your proposal, the config file for Rev.E USB2 camera shield can be found from: 
http://arducam.com/downloads/modules/IMX298/config/USB2-REVE/ 
Now only maximum resolution is available, we will port the other resolutions very soon.
Our camera is not UVC compliance, you have to use our SDK for using the camera.
Anyway, we definetly can make it work, please contact us directly for more information by  support@arducam.com

AmrEzzatIbrahim, 2021-12-16:
@ArduCAM I have the REV.C cam and couldn't find the config too , also the [link](http://arducam.com/downloads/modules/IMX298/config/USB2-REVE/) above isn't working form me , neither are any of the parent pages

zorrobyte, 2022-05-10:
Ended up losing the camera during a move, issue can be considered resolved

closed 2022-05-10 by zorrobyte