ArduCAM/ArduCAM_USB_Camera_Shield #124: Raspberry Pi Camera Module V2 with USB Camera shield

ziwenjiang, 2019-10-05:
Hello, I recently bought the USB 3.0 Camera shield and the corresponding MIPI adapter so I can connect the Pi camera onto an Nvidia AGX through USB port. I have two questions about the current Jetson TX1/2 SDK:
1. In the python streaming demo, it looks like I need to pass a camera configuration json file. However, it seems like IMX219's configuration file is not available. Can you tell me if I have to pass the configuration file for the camera shield to work? Or once the pi camera connects to the shield, it will be automatically recognized as a USB camera?
2. Is there a way to read the RAW image file from the camera through ArduCAM SDK?
Thanks for the help!

PinkSnake, 2019-10-29:
Hello,
Same question with the OV2311 :)
Thx for the SDK!

ziwenjiang, 2019-10-30:
I think the best way to get a fast response is to contact them directly by email, support@arducam.com.

jpwallacepc, 2019-11-06:
I had a similar issue regarding another OV9281 config file, and I had help pointing me to a bunch of other camera config's in the Windows folder.  
See my closed issue here; https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/issues/128

They are in .cfg format but using the Parser tool listed, you can convert them to .yml or .json file if needed.

Hope this helps :)