ArduCAM/ArduCAM_USB_Camera_Shield #128: OV9281 configuration file for cpp_config

jpwallacepc, 2019-10-30:
I'm trying to get two OV9281's working with the Rev E usb shield, but I can't find any config files for them. Does anybody know a way around this, or is it possible to have the cpp_config folder updated.

Or maybe the V4l2 drivers could work with a command other than path/to/.yml?

Cheers for any help

glddiv, 2019-10-31:
Hi @jpwallacepc 
Please check this: https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/issues/119#issuecomment-522391323 and the configuration file is here: [Rev E usb shield config files](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Windows/GUI/USBTest/x64/Release/Config/USB2.0_UC-391%20Rev.E/MIPI)

jpwallacepc, 2019-10-31:
Hey thanks. I'll take a look at playing with all this in a couple of days. I didn't mention that I'm on a RPI Zero but maybe it doesn't make to much difference. Cheers for hooking me up.

jpwallacepc, 2019-11-06:
Thanks again. It's taken me a bit of time as I've had a few other issues, but in the end the conversion works perfectly.

closed 2019-11-06 by jpwallacepc