ArduCAM/ArduCAM_USB_Camera_Shield #131: MT9J001 Lines

iMhack, 2019-12-03:
Hello,
I'm using the MT9J001 breakboard with the UC-425Rev.C module.
I installed this on a clean install Raspberry Pi 4: 

`sudo apt-get update && sudo apt-get --assume-yes dist-upgrade && sudo apt-get install git && git clone https://github.com/ArduCAM/arducam_config_parser.git &&  cd arducam_config_parser &&  make clean && make && make install && cd ~ && wget -O libusb-1.0.22.tar.bz2  https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.22/libusb-1.0.22.tar.bz2/download && tar -jxvf   libusb-1.0.22.tar.bz2 && cd libusb-1.0.22 && ./configure --disable-udev && sudo make install && sudo apt-get --assume-yes install libopencv-dev g++-4.9 && cd ~ && git clone https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield.git && cd ArduCAM_USB_Camera_Shield/RaspberryPi/Cpp/Streaming_demo/ && make clean && make install-sdk && make && wget https://arducam.com/downloads/modules/industrial/MT9J003/MT9J001_10MP_8bit.cfg && wget https://arducam.com/downloads/modules/industrial/MT9J003/MT9J001_1080p_8bit.cfg && wget https://arducam.com/downloads/modules/industrial/MT9J003/MT9J001_720p_8bit.cfg && wget https://arducam.com/downloads/modules/industrial/MT9J003/MT9J001_VGA_8bit.cfg && sudo ./ArduCam_Demo MT9J001_VGA_8bit.cfg `

I did some tests and here are the results 
[image]
[image]
[image]
[image]

Few questions :

1. What can I do to avoid the lines shown on VGA, 720 and 1080 ?
2. Can I found some documentation for setting-up the sensor ?
3. What FPS can I get in different configurations ? What are the limiting factors ?

ArduCAM, 2019-12-04:
We now moved our support to our forum, please resent the questions to https://www.arducam.com/forums/ 
Thank you.

iMhack, 2019-12-04:
Hello @ArduCAM , I can't create a login on your private website. Can't you answer here ?

UCTRONICS, 2019-12-05:
1. What can I do to avoid the lines shown on VGA, 720 and 1080 ?
A: The low resolution is got form skipping the high resolution, It is due to the skip pix. We can't avoid it. 
2. Can I found some documentation for setting-up the sensor ?
About the sensor's document, we need to require from the local factory.
3. What FPS can I get in different configurations ? What are the limiting factors ?
As normal the lower resolution, you will get the faster frame speed. Many factors will influence the frame speed such as the exposure time .etc.

iMhack, 2019-12-05:
I think that the factory is in China. It's not so local ;-)

iMhack, 2019-12-05:
I found these ref :
https://www.arducam.com/downloads/modules/industrial/MT9J003_RR_A.pdf
https://files.niemo.de/aptina_pdfs/MT9J001_J003_DG.book.pdf