ArduCAM/ArduCAM_USB_Camera_Shield #142: USB3.0 UC-425 Rev.C with IMX477 (High quality camera) not displaying images correctly or too dim.

alsarmie, 2020-09-25:
Hello,
I  recently got the new high-quality camera IMX477 with the USB3 shield. I was expecting to be able to use it straight away as advertised here (sensor and USB shield): https://www.youtube.com/watch?v=8A3walono1k&ab_channel=Arducam   at (0:30-0:37)
First I tried on Windows for a sanity check.
I am able to load the configuraiton files for the IMX477 sensor, but the image is not being displayed or it is super noisy.

[image]

On Linux (on which I am more interested in getting this to work), I am able to run the python ArduCAM demo with the config files, but I am getting very dim and greenish images. I can not seem to find a way to modify this on the documentation. Your support is appreciated. The IMX477 cfg files are the ones  from the latest repo update.

alsarmie, 2020-09-25:
This is the image I get in Linux:
[image]
After a while (5 mins of video streaming), the demo just freezes with this constant output:
('Error capture image, rtn_val = ', 65318)
('Error capture image, rtn_val = ', 65318)
('Error capture image, rtn_val = ', 65318)
('Error capture image, rtn_val = ', 65318)
('Error capture image, rtn_val = ', 65318)
('Error capture image, rtn_val = ', 65318)

keesjan123, 2021-03-19:
I'm having similar issues with the IMX477 and the USB3 shield. The Linux version is underexposed and has whitebalance problems (image looks green-ish). Also, I have no idea how to change any parameters such as exposure, gain, et cetera.

I figured that perhaps I can change the .cfg file myself and tried looking for the IMX477 datasheet, but I could not find a document detailing the register mappings so far.

This makes me wonder: where does ArduCam get their register mapping documentation from? Any help is greatly appreciated.

alsarmie, 2021-03-19:
You can modify gain and exposure by modifying camera registers,
gain:
MX477_ANALOG_GAIN_ADDR_MSB 0x0204
IMX477_ANALOG_GAIN_ADDR_LSB 0x0205
min=0 max=978
exposure:
IMX477_COARSE_INTEG_TIME_ADDR_MSB 0x0202
IMX477_COARSE_INTEG_TIME_ADDR_LSB 0x0203

That is the answer I got from their support. However for the rest of the parameters, only they know how to configure them.  The other errors I saw were due to USB bandwidth from the PC. However, I would advise to be careful with the USB3 shield, the connector is pretty weak and tends to break easily with normal usage :/.