ArduCAM/ArduCAM_USB_Camera_Shield #144: Ubuntu 20 Compatibility

NunoGaioPereira, 2020-10-19:
Hello,

I have noticed from [this post ](https://www.arducam.com/forums/topic/ubuntu-20-04-does-not-include-opencv-3-2-that-is-needed-by-libarducamlib-so-2-0/#post-29396) that the library hasn't been compiled with Ubuntu 20.

I am working on a commercial project and we have acquired and very high number of arducam’s (model AR0134-C).
We acre currently developing a new product that will rely on the arducam for drone navigation.
However, we need to use Ubuntu 20 and it is essential for us to integrate the camera with the ArduCAM USB shield as fast as possible.

Could you please compile the .so file in Ubuntu20 so it is compatible and we can use it in our solution?

Many thanks

ArduCAM, 2020-10-19:
please send an email to support@arducam.com for the support

ArduCAM, 2020-10-20:
Hi @NunoGaioPereira 
Can you describe the problem you encountered in detail?
libArduCamLib.so.2.0.0 should not depend on the system version.

glddiv, 2020-10-20:
Hi @NunoGaioPereira 
Can you describe the problem you encountered in detail?
libArduCamLib.so.2.0.0 should not depend on the system version.

NunoGaioPereira, 2020-10-20:
Hello @glddiv  , thank you for your reply!

I followed the instructions found in [this example](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Linux_x86/Cpp/Streaming_demo).

After running `make install-sdk` and the .so files are copied to /usr/lib.

When I run `make` get the following:
```
g++ ArduCam_Demo.cpp Arducam_SDK/ArduCamLib.h -o ArduCam_Demo -lArduCamLib -lusb-1.0  -lpthread -larducam_config_parser `pkg-config --cflags --libs opencv4` -L. -I. -std=gnu++11 -g 
/usr/bin/ld:/usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libArduCamLib.so: file format not recognized; treating as linker script
/usr/bin/ld:/usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libArduCamLib.so:0: syntax error
collect2: error: ld returned 1 exit status
make: *** [makefile:8: ArduCam_Demo] Error 1
```
Which seems like there could be some incompatibility with the compiler I am using or something wrong with the libArduCamLib.so file.

After running `make install-sdk` file the .so files are copied to /usr/lib, but then it seems to be looking for them in another 

If I run **ls -ltr ./libArduCamLib.so** I obtain:
`-rw-rw-r-- 1 nuno nuno 22 out 19 11:06 ./libArduCamLib.so`
If I run **file ./libArduCamLib.so** I obtain:
`./libArduCamLib.so: ASCII text, with no line terminators`

Specs:
 - Ubuntu 20
 - gcc version: 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04) 
 - libopencv-dev: 4.2.0+dfsg-5
 - libusb: 1.0.23

Could you please help? I could be missing something.

Many thanks,
Nuno

NunoGaioPereira, 2020-10-20:
> Hello @glddiv , thank you for your reply!
> 
> I followed the instructions found in [this example](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Linux_x86/Cpp/Streaming_demo).
> 
> After running `make install-sdk` and the .so files are copied to /usr/lib.
> 
> When I run `make` get the following:
> 
> ```
> g++ ArduCam_Demo.cpp Arducam_SDK/ArduCamLib.h -o ArduCam_Demo -lArduCamLib -lusb-1.0  -lpthread -larducam_config_parser `pkg-config --cflags --libs opencv4` -L. -I. -std=gnu++11 -g 
> /usr/bin/ld:/usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libArduCamLib.so: file format not recognized; treating as linker script
> /usr/bin/ld:/usr/lib/gcc/x86_64-linux-gnu/9/../../../../lib/libArduCamLib.so:0: syntax error
> collect2: error: ld returned 1 exit status
> make: *** [makefile:8: ArduCam_Demo] Error 1
> ```
> 
> Which seems like there could be some incompatibility with the compiler I am using or something wrong with the libArduCamLib.so file.
> 
> After running `make install-sdk` file the .so files are copied to /usr/lib, but then it seems to be looking for them in another 
> 
> If I run **ls -ltr ./libArduCamLib.so** I obtain:
> `-rw-rw-r-- 1 nuno nuno 22 out 19 11:06 ./libArduCamLib.so`
> If I run **file ./libArduCamLib.so** I obtain:
> `./libArduCamLib.so: ASCII text, with no line terminators`
> 
> Specs:
> 
> * Ubuntu 20 x64bits
> * gcc version: 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)
> * libopencv-dev: 4.2.0+dfsg-5
> * libusb: 1.0.23
> 
> Could you please help? I could be missing something.
> 
> Many thanks,
> Nuno

Quick update:
I added the -c flag to the compilation step and I don't have the `file format not recognized; treating as linker script` error anymore.
But the compilation still didn't create an executable, so I ran `chmod +x ArduCam_Demo`.
Howerver, if I run 
`sudo ./ArduCam_Demo` with the path to the config file, I get:
```
./ArduCam_Demo: 1: gpch+014Fo���/0�x?����wPx86-64generic__SSP_STRONG__ 3__DBL_MIN_EXP__ (-1021)__cpp_attributes 200809�__FLT128_MAX_10_EXP__ 4932
                                            __FLT_MIN__ 1.1__GCC_IEC_559_COMPLEX 2__UINT_LEAST8_TYPE__: not found
./ArduCam_Demo: 2: Syntax error: word unexpected (expecting ")")
```
which just seems Like I shouldn't have made in an executable with chmod or there is something wrong with compilation/linker.

I also tested the same example with Python and I made it worked.

Thanks.

chutsu, 2021-05-05:
@NunoGaioPereira Have you been able to resolve this issue?

NunoGaioPereira, 2021-05-09:
Hey @chutsu!
I stopped working with this camera meanwhile but from what I remember I don't think so.

It worked quite well with Python though.

I will close this issue.

closed 2021-05-09 by NunoGaioPereira