ArduCAM/ArduCAM_USB_Camera_Shield #149: issues with AR0134-C with UC391-rev.D shield and external trigger

flaflip, 2020-12-11:
Hi.
Using a AR0134-C with UC391-rev.D shield and the out of the box demo repo..

When I run the external trigger script from a shell I get:
$ sudo python ArduCam_Ext_Trigger_Demo.py /home/pi/anySorting/ArduCAM_USB_Camera_Shield/Config/USB2.0_UC-391_Rev.D/DVP/AR0134/AR0134_RAW_8b_ext_trigger_1280x964.cfg
Found 1 devices
('Index:', 0, 'Serial:', '\xff\xff\xff\xff-\xff\xff\xff\xff-\xff\xff\xff\xff')
initialization done
Serial: ����-����-����
Take picture.
('Take picture fail,ret_val = ', 65318L)
Take picture.
('Take picture fail,ret_val = ', 65318L)
Take picture.
('Take picture fail,ret_val = ', 65318L)
Take picture.
('Take picture fail,ret_val = ', 65318L)
Take picture.
('Take picture fail,ret_val = ', 65318L)
Take picture.

Weird thing is that I did not connect anything to the external trigger. I did check 

2:
When I run the Streaming_demo, I get:
color mode 2
{'emI2cMode': 3,
 'emImageFmtMode': 0,
 'u16Vid': 0,
 'u32CameraType': 0,
 'u32Height': 964,
 'u32I2cAddr': 32,
 'u32Size': 0,
 'u32TransLvl': 0,
 'u32Width': 1280,
 'u8PixelBits': 8,
 'u8PixelBytes': 1,
 'usbType': 0}
Serial: ÿÿÿÿ-ÿÿÿÿ-ÿÿÿÿ
Capture began, rtn_val =  0
Error capture image, rtn_val =  65317
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
Error capture image, rtn_val =  65316
fps: 0 /s

Then an image.. (SO IT WORKS)
I tried lowering the pll_multiplier as stated in https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/issues/105 
and that results in less, but still errors 

All help is appreciated.

glddiv, 2020-12-15:
Hi @flaflip 
> Weird thing is that I did not connect anything to the external trigger. I did check

This is really weird, I cannot reproduce this phenomenon..

You can use Streaming_demo with the AR0134_RAW_8b_ext_trigger_1280x964.cfg configuration file. When it is not triggered, it will cause many 65318 errors, but this does not affect the use.

flaflip, 2020-12-15:
Thanks for your response.

So I tried using the  Streaming demo script with the AR0134_RAW_8b_ext_trigger_1280x964.cfg file.

When I do this, I get indeed a lot of 65318 errors and no 65316 anymore.
But it also does not create any picture.. 

About the setup: I am not using the external trigger I plan to use a 5v trigger sensor on the GPIO on the Raspi that is monitored in the script.
This 5V trigger is not suitable for the external trigger of the camera since that is 3.3 volts; I don't want to break the shield.. And I dislike creating all kinds of hardware conversion for now..

I would be really happy if i can get a stable stream of images from the device. I'll pick the right image with the sensor later :-)

glddiv, 2020-12-22:
Hi @flaflip 
Sorry for the late reply,

> With these on, I get:
AttributeError: module 'ArducamSDK' has no attribute 'Py_ArduCam_registerCtrls'

This error may be due to you not updating the latest SDK.

> Update(1):
> After this, I switched back to the original setup with Streaming demo and AR0134_RAW_8b_1280x964_31fps.cfg
> 
> It immediately starts with:
> color mode 2
> {'emI2cMode': 3,
> 'emImageFmtMode': 0,
> 'u16Vid': 0,
> 'u32CameraType': 0,
> 'u32Height': 964,
> 'u32I2cAddr': 32,
> 'u32Size': 0,
> 'u32TransLvl': 0,
> 'u32Width': 1280,
> 'u8PixelBits': 8,
> 'u8PixelBytes': 1,
> 'usbType': 0}
> Serial: ÿÿÿÿ-ÿÿÿÿ-ÿÿÿÿ
> Capture began, rtn_val = 0
> Error capture image, rtn_val = 65317
> fps: 0 /s
> 
> The first image is also ok.
> But after second image image it starts spitting "Error capture image, rtn_val = 65316" again.

Does the 65316 error keep appearing? Is there a normal picture after this? This error may be caused by insufficient USB bandwidth.

> So I tried using the Streaming demo script with the AR0134_RAW_8b_ext_trigger_1280x964.cfg file.
> 
> When I do this, I get indeed a lot of 65318 errors and no 65316 anymore.
> But it also does not create any picture..

This needs to send an external trigger signal to the camera. Can you provide a hardware connection diagram?

flaflip, 2020-12-29:
hi @glddiv 
thanks for your response.

global update: it is working now. I updated the complete repo and run the code directy from shell. then I get the 65316  only once in the beginning.

I have no need for the external trigger since I am using a different mechanism now. Offtopic question: will it work with a 5v signal or will I destroy the hardware when I do this?

Some learnings that i'd like your opinion about befor we close the topic:
I am trying to make development a bit more convenient so I am using jupyter notebook.
When I use jupyter, python3 and output the images with matplotlib I still get these 65316 errors. and then after 10 or 20 errors I get a normal image..

When I change time.sleep(0.001) of both threads to > 0.09 in the running example I get the same 65316 issue in the console..

So it's some kind of timing issue?

glddiv, 2021-01-05:
Hi @flaflip 

> I have no need for the external trigger since I am using a different mechanism now. Offtopic question: will it work with a 5v signal or will I destroy the hardware when I do this?

Yes, this will most likely damage the hardware.

> When I change time.sleep(0.001) of both threads to > 0.09 in the running example I get the same 65316 issue in the console..

It is best not to add sleep in the capture thread. I have a sample program that wraps the arducam api into a class, you can refer to it.
[arducam_python_example.zip](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/files/5769385/arducam_python_example.zip)

flaflip, 2021-01-05:
Thanks! it works!

closed 2021-01-05 by flaflip