ArduCAM/ArduCAM_USB_Camera_Shield #152: can't display 752 x 480 on MT9V022 with edits to  ArducamCfg.json

coldstart01, 2021-03-08:
Can only toggle 752 x 480 mode in USB Test App with direct input to registers. I try the same setup in another solution and itr only works by editing the .JSON file to 640 x 480 resolution:  is the Arducam MT9V022  752 x 480 really supported?

coldstart01, 2021-03-08:
Not working JSON file:

{
  "camera_param": {
    "emI2cMode": 1,
    "emImageFmtMode": 4,
    "u32Height": 480,
    "u32Width": 640,
    "u32I2cAddr": "0x90",
    "u8PixelBits": 8,
    "u8PixelBytes": 1,
    "u32TransLvl": 64
  },
  "board_param_usb2": [
    [
      "0xD7",
      "0x4600",
      "0x0100",
      "0x01",
      [ "0x05" ]
    ],
    [
      "0xD7",
      "0x4600",
      "0x0200",
      "0x01",
      [ "0x00" ]
    ],
    [
      "0xD7",
      "0x4600",
      "0x0300",
      "0x01",
      [ "0x00" ]
    ]
  ],
  "board_param_usb3": [
    [
      "0xF3",
      "0x0000",
      "0x0000",
      "0x00",
      []
    ],
    [
      "0xF9",
      "0x0000",
      "0x0000",
      "0x00",
      []
    ]
  ],
  "register_param_usb2": [
    [ "0x03", "0x01E0" ],
    [ "0x04", "0x0280" ],
    [ "0x0D", "0x0300" ]
  ],
  "register_param_usb3": [
    [ "0x03", "0x01E0" ],
    [ "0x04", "0x0280" ],
    [ "0x0D", "0x0300" ]
  ]
}

Working copy:
{
  "camera_param": {
    "emI2cMode": 1,
    "emImageFmtMode": 4,
    "u32Height": 480,
    "u32Width": 640,
    "u32I2cAddr": "0x90",
    "u8PixelBits": 8,
    "u8PixelBytes": 1,
    "u32TransLvl": 64
  },
  "board_param_usb2": [
    [
      "0xD7",
      "0x4600",
      "0x0100",
      "0x01",
      [ "0x05" ]
    ],
    [
      "0xD7",
      "0x4600",
      "0x0200",
      "0x01",
      [ "0x00" ]
    ],
    [
      "0xD7",
      "0x4600",
      "0x0300",
      "0x01",
      [ "0x00" ]
    ]
  ],
  "board_param_usb3": [
    [
      "0xF3",
      "0x0000",
      "0x0000",
      "0x00",
      []
    ],
    [
      "0xF9",
      "0x0000",
      "0x0000",
      "0x00",
      []
    ]
  ],
  "register_param_usb2": [
    [ "0x03", "0x01E0" ],
    [ "0x04", "0x0280" ],
    [ "0x0D", "0x0300" ]
  ],
  "register_param_usb3": [
    [ "0x03", "0x01E0" ],
    [ "0x04", "0x0280" ],
    [ "0x0D", "0x0300" ]
  ]
}

coldstart01, 2021-03-08:
Also the 752 x 480 will work in USB Test app when "fit to window" is checked.

chutsu, 2021-04-27:
You need to set the register `0x04` (Window Height for Context A) to `0x2F0`, which is hexidecimal for 752. Your current JSON file you pasted above is setting the register to `0x0240` which is hexidecimal for 680. See page 5 and 9 of the MT9V034 Register Reference for more info. I cannot link you the document because technically the document is confidential and proprietary, but you can obtain it easily.