ArduCAM/ArduCAM_USB_Camera_Shield #170: Windows 10 Cannot find driver

ste-arm, 2021-07-27:
I am following the [windows driver installation instructions](https://www.arducam.com/docs/usb-cameras/windows-driver-installation/) from the Arducam website. I have enabled unsigned drivers. When I plug the camera in, Windows is automatically installing a generic "
[image]
USB Video Device", not showing as unrecognized as in the instructions. when I try and update the driver manually by searching the git hub files "C:\Users\user\ArduCAM_USB_Camera_Shield\Drivers\x64" only the cyusb.inf file is visible and I receive a message saying "The folder you specified doesn't contain a compatible software driver for your device. If the folder contains a driver, make sure it is designed to work with Windows for x64-based systems."

andremessias, 2021-09-24:
Hello @ste-arm !
Did you fixed that problem?

I also get this error following the drive installation instructions. However, my W10 always install the "automatic" windows drivers.
I would like to use the ArdumCam USB Camera Demo because I would like to change the CfgFile to change the default frame size.

I'm building a sollution that as to take a frame/shot after a trigger, but I have to set the frame size ever I take the frame. That means to take more 1,2 seconds in my process.

Can you gime your opinion/feedback?

conrad503, 2021-10-05:
Same problem.  Has anyone found a solution?

WaterCube001, 2021-10-18:
same problem.

KwonHyeonSu, 2022-03-26:
same too

mcix, 2022-05-19:
I have the same issue, has anyone found a solution?

mcix, 2022-05-31:
I was trying to use it with an Arducam CSI-USB UVC Camera Adapter Board, but for that board this application should be used:
https://www.arducam.com/docs/uvc-camera-module/quick-start-with-windows/using-amcap-with-arducam-usb-cameras/