ArduCAM/ArduCAM_USB_Camera_Shield #187: ArduCAM 108MP camera demo produces green shaded images

nrm495, 2022-09-21:
Using the configuration and demo file below, I'm able to capture frames from the camera, but they are green shaded when using color_mode = 0 or 3. Setting color_mode = 1 or 2 produces a red shaded image. How do I get a normal "true" color image (roughly speaking - I understand the exact color depends on the monitor settings, etc.). 

Config file I'm using: 
ArduCAM_USB_Camera_Shield/Config/USB3.0_UC-593 Rev.C/MIPI/ArduCAM_108MP/ArduCAM_108MP_MIPI_2Lane_RAW8_12000x9000_1.4fps.cfg

Demo script I'm using:
ArduCAM_USB_Camera_Shield_Python_Demo/ArduCam_Demo.py

I'm running this in Python3 on Windows 10

anandhakrishnantecholution, 2023-02-06:
Facing the same issue as well.

ben-mallett, 2023-11-11:
Adding a 5 second sleep after the opening of the camera appears to work for me.